google/gf180mcu-pdk #4: Set up some type of CI for this repository

mithro, 2022-07-20:
We should have CI running on this repository. 

@atorkmabrains can you write down the initial plan in this GitHub Issue?

atorkmabrains, 2022-07-20:
Yes

atorkmabrains, 2022-07-21:
Hi @mithro ,

We have about 60% of the regression tests for gf180mcu-pdk built for most of the PDK components. We just need to convert the running of those regression to be as CI. 

@mithro What do you recommend us doing: adding those regression scripts into this main repo or having a separate repo called regression to include all the testing?

mithro, 2022-07-21:
If I understand what you are saying, then please add them directly to this repository.

atorkmabrains, 2022-07-22:
Sounds Great, we will add regression tests as CI. BTW, they take long time to run.

mithro, 2022-07-22:
@atorkmabrains - Taking a long time to run is fine, we have a bunch of ways we can speed that up.

mithro, 2022-07-22:
Related GitHub Issues on other parts of the PDK;
 * https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_sc_mcu9t5v0/issues/2
 * https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_sc_mcu7t5v0/issues/2
 * https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/issues/3
 * https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr/issues/4

Testing related GitHub Issues on the PDK;
 * https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_sc_mcu9t5v0/issues/5
 * https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_sc_mcu7t5v0/issues/4
 * https://github.com/google/globalfoundries-pdk-ip-gf180mcu_fd_ip_sram/issues/2
 * https://github.com/google/globalfoundries-pdk-ip-gf180mcu_fd_ip_sram/issues/3
 * https://github.com/google/globalfoundries-pdk-ip-gf180mcu_fd_ip_sram/issues/4

atorkmabrains, 2022-07-25:
@mithro @mohanad0mohamed is currently working on it.