google/gf180mcu-pdk #30: Add robot to update common files in all submodules

mithro, 2022-07-25:
There are a bunch of things which need to be applied to all the submodules of the gf180mcu-pdk. We should set up a robot which makes sure these files are kept in sync and deployed.

For the F4PGA project their where students at BYU who were working on setting up something like this.

mithro, 2022-07-25:
@jgoeders - Do you know what happens with the project on F4PGA?

mithro, 2022-07-25:
@acomodi / @kgugala -- Do you know what happened with that project at BYU for F4PGA?

jgoeders, 2022-07-25:
@xanjohns - Is this the "common config" project?  What's the status of that?

mithro, 2022-07-25:
@jgoeders - Yes, the "common config" project was what I was thinking about.

xanjohns, 2022-07-26:
@mithro Hello! The common-config project should be able to handle something like this pretty easily with how we have it set up right now.

We would need to:

1.  Add all of the files that need to be kept in sync in a new directory in [symbiflow-common-config](https://github.com/SymbiFlow/symbiflow-common-config)
2. Update the [sync config file](https://github.com/SymbiFlow/symbiflow-common-config/blob/main/.github/sync.yml) with the list of repositories that need to be kept in sync.

I can take care of both of these things, but if someone can point me to what files need to be kept in sync, and which submodules we need to  target (are they the ones [google/gf180mcu-pdk/libraries](https://github.com/google/gf180mcu-pdk/tree/main/libraries)?) that would be great.

xanjohns, 2022-08-16:
@mithro Is there someone that I should consult to answer my questions above?