google/gf180mcu-pdk #42: Update the names of files under `docs/analog/model_parameters/LV/tables_clear`

mithro, 2022-08-02:
The names of the files under [`docs/analog/model_parameters/LV/tables_clear`](https://github.com/google/gf180mcu-pdk/tree/main/docs/analog/model_parameters/LV/tables_clear) should probably match primitive naming?

It might actually just be that the formatting of these file names need to match the other ones in this file? The files which made me think where the following;
 * `docs/analog/model_parameters/LV/tables_clear/19_mos_3p3.csv`
 * `docs/analog/model_parameters/LV/tables_clear/21_mos_6p0.csv`
 * `docs/analog/model_parameters/LV/tables_clear/22_mos_6p0_1.csv`
 * `docs/analog/model_parameters/LV/tables_clear/21_mos_6p0_2.csv`

mithro, 2022-08-02:
I think they might just be `_MOS_` or `_Mos_` -- `_Mos_` doesn't look right, so I'm guessing it `_MOS_` is right?

mohanad0mohamed, 2022-08-04:
Should we stick to the naming convetion of nfet/pfet instead?
I'm suggesting `19_mos_3p3.csv` to be `19_nfet_pfet_03v3.csv`