google/gf180mcu-pdk #65: Liberty file header information is not in the PDK

RTimothyEdwards, 2022-08-23:
## Expected Behavior
`make timing` generates liberty headers and creates the liberty libraries (as it does in the sky130 open PDK)

## Actual Behavior
There is no `make timing`, and there are no liberty file headers.

Probably it is close to sufficient to just copy the timing script from sky130 PDK and add the build recipe to the Makefile.  There is probably some process-dependent stuff in the script that needs to be modified.

@mithro

proppy, 2022-09-14:
re-opening since we still need to add a `make timing` equivalent.

tstrahler, 2022-11-07:
The Liberty file header does not contain

tstrahler, 2022-11-07:
The Liberty files still do not contain any information about the individual cells. This makes the PDK basically unusable when working with commercial synth tools.