google/gf180mcu-pdk #71: Incorrect Cell LEF used to generate the Standard Cells

QuantamHD, 2022-09-01:
## Expected Behavior
The standard cell lef layer names should match the layers defined in the tech LEF.

## Actual Behavior
The TLEF layer names are named ["Metal1", "Metal2"](https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_sc_mcu9t5v0/blob/main/tech/gf180mcu_5LM_1TM_9K_9t_tech.lef#L65)..., but the cell LEFs use an all caps naming convention,[ "METAL1", "METAL2".](https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_sc_mcu9t5v0/blob/main/cells/and3/gf180mcu_fd_sc_mcu9t5v0__and3_1.lef#L68)

It looks like the wrong cell LEF was used to extract the cell definitions. In the upstream PDK there are two cell lef files one ending in  _oa the ending in _mw. The _oa lef file has the all caps naming convention. We should be using the _mw which has the correct layer names.

[image]

QuantamHD, 2022-09-01:
Hi @FaragElsayed2,

I work on Tim's team at Google. I'm just doing a final review of the PDK to find blocking issues.

atorkmabrains, 2022-09-02:
@QuantamHD We will get back to you on Sunday.

mithro, 2022-09-06:
@atorkmabrains - Poke?

atorkmabrains, 2022-09-06:
@mithro Sorry we were stuck in something the last few days. Will get that done tomorrow.

proppy, 2022-09-07:
isn't this fixed by?
https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_sc_mcu9t5v0/pull/15
https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_sc_mcu7t5v0/pull/17

proppy, 2022-09-14:
Marking as fixed since the underlying PR were merged.

closed 2022-09-14 by proppy