google/gf180mcu-pdk #82: Standard cell page in the docs is broken

mithro, 2022-10-17:
## Expected Behavior

The page @ https://gf180mcu-pdk.readthedocs.io/en/latest/digital/standard_cells/standard_cells.html should list the standard cell libraries.

## Actual Behavior

The standard cell libraries are missing currently. See below;
[image]

Version - GlobalFoundries GF180MCU PDK `0.0.0-75-g331e678` documentation

mithro, 2022-10-17:
I think this is being caused by the `submodules: recursive: false` option added @
https://github.com/google/gf180mcu-pdk/pull/80/files#diff-03efc769b870804394632e45d7885272b44c16939517fb31c9d7c614d2ffae57R29

proppy, 2022-10-19:
> I think this is being caused by the submodules: recursive: false option added @

Looks like `false` is the default https://docs.readthedocs.io/en/stable/config-file/v2.html#submodules-recursive, so I don't think that's the culprid (and also https://github.com/google/open-source-pdks doesn't seem to have any nested submodules).

https://github.com/google/gf180mcu-pdk/pull/80/commits/60550f39ec47e36bf13eab7bd1b48d5e5490aa11 introduced a explicit list of submodules which could cause readthedocs to only clone those, @cbalint13 can you comment if that was intentional?

Switching to:
```
submodules:
  include: all
``` 
should allow all submodules to get cloned.

cbalint13, 2022-10-19:
> [60550f3](https://github.com/google/gf180mcu-pdk/commit/60550f39ec47e36bf13eab7bd1b48d5e5490aa11) introduced a explicit list of submodules which could cause readthedocs to only clone those, @cbalint13 can you comment if that was intentional?

@proppy 

I picked up only **that one module** without beeing aware that **before all was picked** behaviour.

* Goal was to make the after **pdf generation** working and we upgrade to **v2**  too.
* The v2 upgrade puzzled us in a way, see:  https://github.com/google/gf180mcu-pdk/pull/80#issuecomment-1270494197 .

proppy, 2022-10-19:
@cbalint13 thanks for the background, can you take a look at #83 and verify it doesn't affect the PDF support you contributed?

cbalint13, 2022-10-19:
> @cbalint13 thanks for the background, can you take a look at #83 and verify it doesn't affect the PDF support you contributed?

@proppy 

PDF are never generated on the open PR only after push is merged (that was also our puzzle).
If you merge it, it will generate it at: https://skywater-pdk.readthedocs.io/_/downloads/en/main/pdf

proppy, 2022-10-20:
Looks like it's still broken even after #85.

```
processing globalfoundriesgf180mcupdk.tex... index digital/Digital digital/standard_cells/standard_cells digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/index digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/setup digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/symbols digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/spec/physical digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/spec/electrical digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/spec/transitions digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/spec/corners digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/addf/gf180mcu_fd_sc_mcu7t5v0__addf_1 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/addf/gf180mcu_fd_sc_mcu7t5v0__addf_2 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/addf/gf180mcu_fd_sc_mcu7t5v0__addf_4 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/addh/gf180mcu_fd_sc_mcu7t5v0__addh_1 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/addh/gf180mcu_fd_sc_mcu7t5v0__addh_2 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/addh/gf180mcu_fd_sc_mcu7t5v0__addh_4 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/and2/gf180mcu_fd_sc_mcu7t5v0__and2_1 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/and2/gf180mcu_fd_sc_mcu7t5v0__and2_2 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/and2/gf180mcu_fd_sc_mcu7t5v0__and2_4 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/and3/gf180mcu_fd_sc_mcu7t5v0__and3_1 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/and3/gf180mcu_fd_sc_mcu7t5v0__and3_2 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/and3/gf180mcu_fd_sc_mcu7t5v0__and3_4 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/and4/gf180mcu_fd_sc_mcu7t5v0__and4_1 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/and4/gf180mcu_fd_sc_mcu7t5v0__and4_2 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/and4/gf180mcu_fd_sc_mcu7t5v0__and4_4 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/antenna/gf180mcu_fd_sc_mcu7t5v0__antenna digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/aoi21/gf180mcu_fd_sc_mcu7t5v0__aoi21_1 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/aoi21/gf180mcu_fd_sc_mcu7t5v0__aoi21_2 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/aoi21/gf180mcu_fd_sc_mcu7t5v0__aoi21_4 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/aoi211/gf180mcu_fd_sc_mcu7t5v0__aoi211_1 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/aoi211/gf180mcu_fd_sc_mcu7t5v0__aoi211_2 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/aoi211/gf180mcu_fd_sc_mcu7t5v0__aoi211_4 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/aoi22/gf180mcu_fd_sc_mcu7t5v0__aoi22_1 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/aoi22/gf180mcu_fd_sc_mcu7t5v0__aoi22_2 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/aoi22/gf180mcu_fd_sc_mcu7t5v0__aoi22_4 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/aoi221/gf180mcu_fd_sc_mcu7t5v0__aoi221_1 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/aoi221/gf180mcu_fd_sc_mcu7t5v0__aoi221_2 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/aoi221/gf180mcu_fd_sc_mcu7t5v0__aoi221_4 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/aoi222/gf180mcu_fd_sc_mcu7t5v0__aoi222_1 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/aoi222/gf180mcu_fd_sc_mcu7t5v0__aoi222_2 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/aoi222/gf180mcu_fd_sc_mcu7t5v0__aoi222_4 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/buf/gf180mcu_fd_sc_mcu7t5v0__buf_1 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/buf/gf180mcu_fd_sc_mcu7t5v0__buf_12 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/buf/gf180mcu_fd_sc_mcu7t5v0__buf_16 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/buf/gf180mcu_fd_sc_mcu7t5v0__buf_2 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/buf/gf180mcu_fd_sc_mcu7t5v0__buf_20 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/buf/gf180mcu_fd_sc_mcu7t5v0__buf_3 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/buf/gf180mcu_fd_sc_mcu7t5v0__buf_4 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/buf/gf180mcu_fd_sc_mcu7t5v0__buf_8 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/bufz/gf180mcu_fd_sc_mcu7t5v0__bufz_1 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/bufz/gf180mcu_fd_sc_mcu7t5v0__bufz_12 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/bufz/gf180mcu_fd_sc_mcu7t5v0__bufz_16 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/bufz/gf180mcu_fd_sc_mcu7t5v0__bufz_2 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/bufz/gf180mcu_fd_sc_mcu7t5v0__bufz_3 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/bufz/gf180mcu_fd_sc_mcu7t5v0__bufz_4 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/bufz/gf180mcu_fd_sc_mcu7t5v0__bufz_8 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/clkbuf/gf180mcu_fd_sc_mcu7t5v0__clkbuf_1 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/clkbuf/gf180mcu_fd_sc_mcu7t5v0__clkbuf_12 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/clkbuf/gf180mcu_fd_sc_mcu7t5v0__clkbuf_16 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/clkbuf/gf180mcu_fd_sc_mcu7t5v0__clkbuf_2 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/clkbuf/gf180mcu_fd_sc_mcu7t5v0__clkbuf_20 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/clkbuf/gf180mcu_fd_sc_mcu7t5v0__clkbuf_3 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/clkbuf/gf180mcu_fd_sc_mcu7t5v0__clkbuf_4 digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/clkbuf/gf180mcu_fd_sc_mcu7t5v0__clkbuf_8  [... cut ...] rce-pdks/contributing open-source-pdks/code-of-conduct 
resolving references...
failed

Extension error:
convert exited with error:
[stderr]
b"convert-im6.q16: must specify image size `/tmp/magick-5973yVPmOKCzxrv' @ error/mvg.c/ReadMVGImage/185.\nconvert-im6.q16: no images defined `/home/docs/checkouts/readthedocs.org/user_builds/gf180mcu-pdk/checkouts/latest/docs/_build/doctrees/images/gf180mcu_fd_sc_mcu7t5v0__endcap.schematic.png' @ error/convert.c/ConvertImageCommand/3258.\n"
[stdout]
b''
```

closed 2022-10-20 by proppy

proppy, 2022-10-20:
Fixed: https://gf180mcu-pdk.readthedocs.io/en/latest/digital/Digital.html

PR #83 (2022-10-19): readthedocs: include all submodules

Fixes #82

TEST=https://gf180mcu-pdk--83.org.readthedocs.build/en/83/digital/standard_cells/gf180mcu_fd_sc_mcu7t5v0/cells/buf/gf180mcu_fd_sc_mcu7t5v0__buf_20.html
PR #87 (2022-10-20): doc: switch to sphinxcontrib.rsvgconverter
Apparently the pdf build failure (see https://github.com/google/gf180mcu-pdk/issues/82#issuecomment-1284785504) makes the final RTD documentation push fails.

Fixes #82 (hopefully)

Also enabled `PDF build` in https://readthedocs.org/dashboard/gf180mcu-pdk/advanced/
[image]

TEST=https://gf180mcu-pdk--87.org.readthedocs.build/en/87/digital/Digital.html

/cc @cbalint13 @mkkassem