google/gf180mcu-pdk #88: MOSFET Multipliers not accounted for, in SPICE Simulation

nataraj-pinakapani, 2022-10-23:
## Expected Behavior
For a MOSFET, when m=2 is specified in the SPICE Netlist, the drain current must be almost twice compared to m=1
In the given SPICE netlist I(VD2) must be twice of I(VD1)

## Actual Behavior
I(VD2) and I(VD1) are almost same. This means, the drain current remains the same irrespective of the value of m. 

## Steps to Reproduce the Problem
SPICE Netlist
---------------------
** sch_path: /home/nataraj/projects/designmyic/cad/gf180mcu_invoke/xschem/sch_lib/nmos_3p3_test.sch
**.subckt nmos_3p3_test
XM1 D G GND GND nmos_3p3 L=1u W=1u nf=1 ad='int((nf+1)/2) * W/nf * 0.18u' as='int((nf+2)/2) * W/nf * 0.18u'
+ pd='2*int((nf+1)/2) * (W/nf + 0.18u)' ps='2*int((nf+2)/2) * (W/nf + 0.18u)' nrd='0.18u / W' nrs='0.18u / W'
+ sa=0 sb=0 sd=0 m=1
XM2 D2 G GND GND nmos_3p3 L=1u W=1u nf=2 ad='int((nf+1)/2) * W/nf * 0.18u' as='int((nf+2)/2) * W/nf * 0.18u'
+ pd='2*int((nf+1)/2) * (W/nf + 0.18u)' ps='2*int((nf+2)/2) * (W/nf + 0.18u)' nrd='0.18u / W' nrs='0.18u / W'
+ sa=0 sb=0 sd=0 m=2
V1 G GND 2
.save i(v1)
VD1 D GND 3.3
.save i(vd1)
VD2 D2 GND 3.3
.save i(vd2)
**** begin user architecture code

.include /home/nataraj/projects/designmyic/cad/pdk/share/pdk/gf180mcuC/libs.tech/ngspice/design.spice
.lib /home/nataraj/projects/designmyic/cad/pdk/share/pdk/gf180mcuC/libs.tech/ngspice/sm141064.spice typical

.control
save all
op
print i(vd1)
print i(vd2)
.endc

**** end user architecture code
**.ends
.GLOBAL GND
.end

## Specifications

- Version: GF180MCU PDK Installed using Open PDK Version 1.0.348
- Simulator: ngspice version 37
- Platform: Ubuntu 22.04.5

nataraj-pinakapani, 2022-10-23:
[image]

msaligane, 2023-07-19:
Any updates on this issue?

@proppy @mithro

d-m-bailey, 2023-07-19:
@nataraj-pinakapani 
My understanding is that
XM1 has L=1u W=1u nf=1 m=1 which corresponds to one physical device with L=1u W=1u 
XM2 has L=1u W=1u **nf=2** m=2 which corresponds to 4 physical devices each with L=1u W=0.5u

Do you get the same results with
L=1u W=1u **nf=1** m=2 which corresponds to 2 physical devices each with L=1u W=1u

d-m-bailey, 2023-07-20:
@ChrisZonghaoLi Great! 

What was the resolution?
- a fix to ngspice?
- a change to netlist parameters?
- simulation parameters?
- fix to the mos models?

ChrisZonghaoLi, 2023-07-20:
> @ChrisZonghaoLi Great!
> 
> What was the resolution?
> 
> * a fix to ngspice?
> * a change to netlist parameters?
> * simulation parameters?
> * fix to the mos models?

I think the original issue might be the parameter `m` was not declared in the spice model somehow. According to @atorkmabrains the new repo (https://github.com/mabrains/globalfoundries-pdk-libs-gf180mcu_fd_pr/tree/main/models) has the issue addressed.

d-m-bailey, 2023-07-20:
Thanks @ChrisZonghaoLi. 
@proppy @mithro Do you want to close this since it appears to be resolved?

FaragElsayed2, 2023-08-16:
@atorkmabrains Yes

atorkmabrains, 2023-08-16:
@FaragElsayed2 Link to PR?

FaragElsayed2, 2023-08-16:
@atorkmabrains 
https://github.com/mabrains/globalfoundries-pdk-libs-gf180mcu_fd_pr/pull/152

atorkmabrains, 2023-08-17:
@d-m-bailey @ChrisZonghaoLi Closing this ticket as it’s already addressed.

closed 2023-08-17 by atorkmabrains