google/gf180mcu-pdk #89: How to use gf180mcu-pdk into openlane flow?

visionvlsi, 2022-11-07:
# How to use gf180mcu-pdk into openlane flow?

mithro, 2022-11-07:
@proppy said the following on the [gf180mcu-pdk-users mailing list](https://groups.google.com/g/gf180mcu-pdk-users/c/wcC9WnVBuWc/m/D08q03tiAwAJ);

> You should be able to use OpenLane with GF180MCU by setting the following environment variables:
> ```
> export PDK_FAMILY=gf180mcu # set environment variable for volare
> export PDK=gf180mcuC # set environment variable for the rest of the toolchain
> ```
> prior to following OpenLane installation instructions: https://github.com/The-OpenROAD-Project/OpenLane#installation-the-short-version
> 
> See also https://caravel-user-project.readthedocs.io/en/latest/ for instructions on how to set up the caravel_user_project environment to submit a project to the recently announced GF-MPW0 Open MPW shuttle:
> https://opensource.googleblog.com/2022/10/announcing-globalfoundries-open-mpw-shuttle-program.html

visionvlsi, 2022-11-07:
I had tried what you had said 

export PDK_FAMILY=gf180mcu # set environment variable for volare
export PDK=gf180mcuC # set environment variable for the rest of the toolchain

and then I tried,  make mount  as been said here to run openlane

https://github.com/efabless/openlane#running-openlane

But, it is still taking sky130 pdk. 

Please help.

vijayank88, 2022-11-09:
@visionvlsi 
Can you provide path where you installed gf180 PDK?
Using in caravel flow or directly in OpenLane flow?

Use following command before running design:
```
export PDK_ROOT=<path to gf180 PDK>
export PDK=gf180mcuC
```

proppy, 2022-11-09:
> I had tried what you had said
> export PDK_FAMILY=gf180mcu # set environment variable for volare
> export PDK=gf180mcuC # set environment variable for the rest of the toolchain

did you run `make` and `make test`, before `make mount`. This will ensure the environment (including the PDK) is set appropriatly as documented in https://github.com/The-OpenROAD-Project/OpenLane#installation-the-short-version.

visionvlsi, 2022-11-10:
> did you run `make` and `make test`, before `make mount`. This will ensure the environment (including the PDK) is set appropriatly as documented in [...]

Let me present what I had done : 

1. I had installed gf180 by following the link 

https://github.com/google/gf180mcu-pdk#building-the-documentation

2. and then before I run make mount I did the following 

i. export PDK_FAMILY=gf180mcu 
ii. export PDK=gf180mcuC [ Here I had tried with path as well like export PDK=<path where I had Installed gf180 by following step 1]
iii. Then I did make mount

proppy, 2022-11-10:
Did you try to run:
```
make
make test
```
After setting the environment variables and before running `make mount`?

visionvlsi, 2022-11-10:
> 

Yes I did that. Let me clearly explain what am I doing 

Firstly I am installing gf180 by following instructions given at the following link 

https://github.com/google/gf180mcu-pdk#building-the-documentation

and then doing the following 

1. git clone https://github.com/efabless/OpenLane.git
2. cd OpenLane
3. export PDK_FAMILY=gf180mcu
4. export PDK=gf180mcuC [ Here I had tried with path as well like export PDK=<path where I had Installed gf180 by following step 1]
5. sudo make

______
After the 5th step what I notice is 

Found version 41c0908b47130d5675ff84 common.py:230
           84255b43f66463a7d6 in                             
           ./dependencies/tool_metadata.yml.                 
Version 41c0908b47130d5675ff8484255b43f66463a7d6 **enabled for 
the sky130 PDK.**

vijayank88, 2022-11-11:
@visionvlsi 
You're cloning wrong repo cause for the setup failure.
Follow below steps:

1. git clone https://github.com/The-OpenROAD-Project/OpenLane.git
2. cd OpenLane
3. export PDK_FAMILY=gf180mcu
4. export PDK=gf180mcuC
5. make
6. make test

visionvlsi, 2022-11-12:
@vijayank88

Thank you for correcting me. You are right, I was cloning the wrong repository. But one thing, after doing the following two steps mentioned by you
1. git clone https://github.com/The-OpenROAD-Project/OpenLane.git
2. cd OpenLane

From step number 3 onwards I have to do those commands as root user then only everything is working fine. If done as a normal user then it is not working!!!

Is that how it works?

proppy, 2022-11-12:
> Is that how it works?

Yes and with those instructions you should need to build the PDK separately like you commented earlier. It gets installed as part of the OpenLane make execution.

closed 2022-11-12 by proppy