google/gf180mcu-pdk #101: Standard cell reference missing for DLATCH in gf180 PDK

0616ygh, 2022-11-30:
## Expected Behavior
1. The latch cells can be found in the PDK by Yosys+ABC.
2. The synthesis, floorplan, placement, and routing steps can be successfully completed by OpenLane with GF180 PDK.
## Actual Behavior
1. ABC can't find latch cells in GF180 PDK and you can find  cells named "$_DLTACH_N_"  in synthesis report.
2. Floorplanning failed for module "$_DLTACH_N_" not found in "merged.nom.lef", and check whether EXTRA_LEFS is set appropriately.

## Steps to Reproduce the Problem

1. git clone https://github.com/0616ygh/rioschip2.git
2. cd rioschip2
3. source env.sh
4. make setup
5. cd openlane 
6. make green

## Specifications

- OpenLane Version: OpenLane daae2154590cf20e0c20b77e3fc02b6526ad09af
- PDK Version: open_pdks 0059588eebfc704681dc2368bd1d33d96281d10f
- Platform: Ubuntu 20.04
[image]

0616ygh, 2022-11-30:
Current solution：
I add a latch_map.v in "_pdks/gf180mcuC/libs.tech/openlane/gf180mcu_fd_sc_mcu7t5v0/_":
```
module \$_DLATCH_P_ (input E, input D, output Q);
  gf180mcu_fd_sc_mcu7t5v0__latq_1 _TECHMAP_DLATCH_P (
    .D(D),
    .Q(Q),
    .GATE(E)
  );
endmodule

module \$_DLATCH_N_ (input E, input D, output Q);
  gf180mcu_fd_sc_mcu7t5v0__latrnq_1 _TECHMAP_DLATCH_N (
    .D(D),
    .Q(Q),
    .GATE_N(E)
  );
endmodule
```

At present, I can successfully pass the floorplan. Is this the right addition? thank you.

QuantamHD, 2022-11-30:
Thanks for the suggestion. I'll have to check what Yosys is looking for when it comes to auto mapping latches. It might be as simple as the liberty file not putting the latch function in quotes

QuantamHD, 2022-11-30:
Looks like yosys might not have support for latches https://github.com/YosysHQ/yosys/issues/1221

0616ygh, 2022-11-30:
Yes, I agree with you very much. This is not supported by Yosys.
In skywater130 pdk, I can find the mapping files in  the`sky130A/libs.tech/openlane/sky130_fd_sc_hd` （I use sky130_fd_sc_hd as an example）: such as latch_map.v,mux_map.v, etc.
So I think in the GF180PDK, it may also be necessary to provide themapping files?  This will be very helpful. Thank you.

QuantamHD, 2022-11-30:
Agreed. @RTimothyEdwards should this be added to the openpdks tech mapping file?