google/gf180mcu-pdk #104: CLASS options wrong in LEF files

thesourcerer8, 2022-12-04:
## Expected Behavior
The LEF files should have lines like "CLASS core ;" where the CLASS parameter has just one option which defines the class.

## Actual Behavior
Several LEF files have several options like e.g.
  CLASS core ;
  CLASS core gf180mcu_fd_sc_mcu9t5v0__tiehIGH ;
  CLASS core gf180mcu_fd_sc_mcu9t5v0__tielOW ;
  CLASS core SPACER ;
  CLASS core WELLTAP ;
  CLASS gf180mcu_fd_sc_mcu9t5v0__endcap PRE ;
which is definitely wrong according to the LEF fileformat specification

QuantamHD, 2022-12-04:
Certainly some of these are problematic, but 

CLASS core SPACER ;
CLASS core WELLTAP ;

are totally valid. See grammar from the LEF/DEF spec. https://www.ispd.cc/contests/18/lefdefref.pdf page 172

[image]

RTimothyEdwards, 2022-12-04:
And +1 to Ethan.  Philipp:  Make sure you're looking at the 5.7 or 5.8 spec.  The requirement of a single value for CLASS is from 5.4 or something;  not exactly sure when they added the optional 2nd argument.

But clearly these issues come from a sloppy application of string replacement, where the original cell names "TIEL", "TIEH", "ENDCAP", and "ANTENNA" (the last one noted by Philipp in a later issue) are cell names that are replaced by the open PDK equivalents, but they are also valid LEF keywords.

QuantamHD, 2022-12-04:
@RTimothyEdwards or @thesourcerer8 I'm not near a keyboard where I could make this change would either of you mind making a PR to fix this?

RTimothyEdwards, 2022-12-04:
@QuantamHD :  These fixes have already been made on the efabless fork.  You need to get a pull request from our fork.

mithro, 2022-12-04:
Does https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_sc_mcu9t5v0/pull/23 look right?

mithro, 2022-12-05:
As far as I can tell, this didn't affect the `mcu7t5v0` library.
```
tansell@tansell-glapstation:~/github/google/globalfoundries-pdk-libs-gf180mcu_fd_sc_mcu7t5v0$ grep -R 'CLASS' | grep -v 'core ;' | grep -v 'core SPACER ;'
cells/tieh/gf180mcu_fd_sc_mcu7t5v0__tieh.lef:  CLASS core TIEHIGH ;
cells/endcap/gf180mcu_fd_sc_mcu7t5v0__endcap.lef:  CLASS ENDCAP PRE ;
cells/antenna/gf180mcu_fd_sc_mcu7t5v0__antenna.lef:  CLASS core ANTENNACELL ;
cells/filltie/gf180mcu_fd_sc_mcu7t5v0__filltie.lef:  CLASS core WELLTAP ;
cells/tiel/gf180mcu_fd_sc_mcu7t5v0__tiel.lef:  CLASS core TIELOW ;
```

closed 2022-12-05 by QuantamHD