google/gf180mcu-pdk #106: Wrong commands in LEF file

thesourcerer8, 2022-12-04:
## Expected Behavior
The LEF files should be conforming to the fileformat specification

## Actual Behavior
There are wrong commands in the LEF files:
[image]

thesourcerer8, 2022-12-04:
gf180mcu_fd_sc_mcu9t5v0__antennaDIFFAREA should be ANTENNADIFFAREA instead.

thesourcerer8, 2022-12-04:
Somehow ANTENNA was replaced by with the antenna cell name, perhaps a TCL scope variable override issue?

QuantamHD, 2022-12-04:
Nice catch, we'll get a fix in.

mithro, 2022-12-05:
I believe this is fixed by https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_sc_mcu9t5v0/pull/23

umarcor, 2022-12-05:
https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_sc_mcu9t5v0/pull/23 was merged. Can this be closed?

thesourcerer8, 2022-12-28:
Can I get these changes into the gfmpw-0d Caravel branch, or will this go into a different branch (which one?), or is there another way to get this fix into a repo with gfmpw-0d ?

QuantamHD, 2022-12-28:
@jeffdi @RTimothyEdwards

RTimothyEdwards, 2022-12-29:
This fix has been in open_pdks since three weeks ago, and by extension should be in volare.  It does not specifically "go into" the caravel repository, although the various commit numbers recorded for the library repositories should all be up-to-date.