google/gf180mcu-pdk #113: Missing design rules?

mguthaus, 2023-01-17:
This page refers to a "table below" but there is no such table:

https://gf180mcu-pdk.readthedocs.io/en/latest/physical_verification/design_manual/drm_07.html

Are there design rules anywhere? I can't seem to find them anywhere in "physical verification".

mguthaus, 2023-01-17:
These are linked from an odd menu on the left. User error.

atorkmabrains, 2023-01-17:
@mguthaus I would really appreciate to get your feedback, how to make it more intuitive. Any ideas?

mguthaus, 2023-01-17:
I updated my initial comment and the topic.

mguthaus, 2023-01-17:
Every time a page is reloaded the expandable menu on the left also resets and closes. This makes it very difficult to navigate.

atorkmabrains, 2023-01-18:
Thanks @mguthaus

thesourcerer8, 2023-02-19:
Perhaps it would also make sense to move all the DRC rules upwards in the hierarchy so that it isn't hidden so deep?